jens62/geberit-aquaclean #2: Home Assistant integration

Niek, 2025-03-21:
Great work @jens62!

Did you think about making a Home Assistant integration (just a thin wrapper on top of your code)? If not, I might give it a try in the next weeks.

Scholdan, 2025-03-21:
Maybe esphome could be an option.

jens62, 2025-03-21:
> Did you think about making a Home Assistant integration

I rarely work with homeassistant because I have been using openHAB for about 12 (?) years.
Can't homeassistant access mqtt?
I have also implemented my openHAB integration by accessing mqtt.
To be honest: I don't want to deal with a homeassistant integration at the moment.

😞

I plan to release my [owntone](https://github.com/owntone/owntone-server) integration with the corresponding openHAB rules in mid-April.
I have used this to implement audio output via Apple Homepod and Google Home in case of events triggered by the AquaClean.

janbuijs91, 2025-05-27:
> Great work [@jens62](https://github.com/jens62)!
> 
> Did you think about making a Home Assistant integration (just a thin wrapper on top of your code)? If not, I might give it a try in the next weeks.

Hi Niek, I was wondering if you did make this into a HA integration?

Kiana321, 2026-02-06:
Hello, I have a Geberit Aqua Clean Maira too and Home Assistant, how we can integrate at Home Assistant please ?

jens62, 2026-02-09:
Hi everyone! 👋

Good news - a complete Home Assistant integration is now available! 🎉

As @jens62 mentioned earlier, the integration uses MQTT (no custom Home Assistant component needed). This approach is simple, reliable, and follows the same pattern as the openHAB integration.

## 📚 Documentation

Complete setup guide and configuration files are now in the repository:
https://github.com/jens62/geberit-aquaclean#home-assistant-integration

## ✨ What's included

- **MQTT Configuration**: Pre-configured sensors, binary sensors, and switch for `configuration.yaml`
- **Dashboard Cards**: Two options (simple and advanced with custom icons)
- **Setup Guide**: Complete installation instructions with troubleshooting
- **Screenshot**: Visual reference of the integration in action

## 🚀 Quick Start

1. Ensure the AquaClean console application is running and publishing to your MQTT broker
2. Navigate to the [`homeassistant/`](https://github.com/jens62/geberit-aquaclean/tree/main/homeassistant) directory
3. Follow the setup instructions in the [SETUP_GUIDE.md](https://github.com/jens62/geberit-aquaclean/blob/main/homeassistant/SETUP_GUIDE.md)

## 🎯 Features

- Monitor user sitting, anal shower, lady shower, and dryer status
- Control toilet lid position
- View device information (SAP number, serial number, production date, etc.)
- Visual dashboard with optional custom SVG icons

@Kiana321 This should help you get started with integrating your Geberit AquaClean Mera into Home Assistant!

Let me know if you have any questions or run into issues with the setup.

mundschenk-at, 2026-02-09:
Unfortunately, this requires you to set up Linux device near your toilet (instead of just an ESP32 BT proxy with the application logic running in HA).

jens62, 2026-02-11:
> [@jens62](https://github.com/jens62) It's totally your prerogative to not want to implement it (especially if you don't use Home Assistant yourself), but the three reasons listed in that LLM text are [solved problems](https://www.home-assistant.io/integrations/bluetooth/#remote-adapters-bluetooth-proxies) in HA.

I still like my agnostic approach and the event driven integration thru mqtt. However, I can also understand your argument 100%. I also enjoy learning new things and will therefore take up your suggestion. So is [ESP32-POE-ISO](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) the hardware of choice, or do you recommend something else?